sinara-hw/Laser_Module #5: Wrong mini DP pinout

filipswit, 2023-02-01:
As is mentioned: https://github.com/sinara-hw/Driver/issues/4, mDP cable has source and sink side, but unfortunatley it looks different than in design, should be like in the table:

  | Driver |   | Laser module | should be
-- | -- | -- | -- | --
1 | GND |   | GND |  
2 | P12V |   | P12V |  
3 | QSPI_IO1_P | Lane0p | QSPI_NCS_P | lane3n
4 | N12V |   | N12V |  
5 | QSPI_IO_N | Lane0n | QSPI_NCS_N | Lane3p
6 | NC |   | NC |  
7 | GND |   | GND |  
8 | GND |   | GND |  
9 | QSPI_IO0_P | Lane1p | QSPI_IO0_P | Lane2n
10 | QSPI_NCS_P | Lane3p | QSPI_IO1_P | Lane0n
11 | QSPI_IO0_N | Lane1n | QSPI_IO0_N | Lane2p
12 | QSPI_NCS_N | Lane3n | QSPI_IO1_N | Lane0p
13 | GND |   | GND |  
14 | GND |   | GND |  
15 | QSPI_CLK_P | Lane2p | QSPI_CLK_P | Lane1n
16 | PC3_ADC_P |   | PC3_ADC3_P |  
17 | QSPI_CLK_N | Lane2n | QSPI_CLK_N | Lane1p
18 | PC2_ADC3_N |   | PC2_ADC3_N |  
19 | GND |   | GND |  
20 | power |   | NC |  

Another thing is that mDP cables usually don't have wired pin no 20.

gkasprow, 2023-02-01:
I'm pretty sure I swapped the pins already...

gkasprow, 2023-02-01:
I took pinout from wikipedia...

filipswit, 2023-04-28:
To summarize these are two bugs:

1) Swapping Diff lines
[image]

2) Connecting P3V3_PD to P3V3s